Seeed-Studio/ArduinoCore-mbed #8: Spi Flash write read demo is not compiling

nobodyman1, 2022-05-25:
[flash_write_read.ino](https://github.com/Seeed-Studio/ArduinoCore-mbed/blob/master/libraries/Spi_Flash/example/flash_write_read/flash_write_read.ino) is not compiling.

`QSPIFBlockDevice.h: No such file or directory`

If I change the include to the full path
`#include "x:\arduino-1.8.19\portable\packages\Seeeduino\hardware\nrf52\2.6.1\cores\arduino\mbed\storage\blockdevice\COMPONENT_QSPIF\include\QSPIF\QSPIFBlockDevice.h"`
a lot of errors are shown, e.g.
`x:\arduino-1.8.19\portable\packages\Seeeduino\hardware\nrf52\2.6.1\cores\arduino\mbed\storage\blockdevice\COMPONENT_QSPIF\include\QSPIF\QSPIFBlockDevice.h:266:5: error: 'qspi_status_t' does not name a type; did you mean 'spi_inst_t'?`

elidonner, 2022-06-03:
If you are using the Arduino IDE, make sure you have the correct board selected. This solved the issue for me.

elidonner, 2022-07-28:
With the update to 2.7.2, I am now also getting this issue.

bradocchs, 2023-02-25:
I'm also trying to run this example but getting this error:
`Compilation error: 'QSPI_FLASH1_IO0' was not declared in this scope`
I'm using version 2.9.1 of the mbed ArduinoCore.

Ideas?

josephduchesne, 2023-11-09:
@bigbearishappy the new build you did here: https://github.com/arduino/ArduinoCore-mbed/commit/9f627f62683f55172d0abd064b5ec3b451030899 seems to have been reverted. The latest 2.9.2 version has the headers for QSPI, but libmbed.a doesn't have the QSPI symbols. I swapped in the version from that commit last year and it fixed the problem (and probably broke other things).

closed 2024-10-15 by baorepo