Seeed-Studio/ArduinoCore-mbed #13: XIAO_BLE_Sense(mbed 2.7.2) battery voltage monitor analogRead(P0_31) does not work

msfujino, 2022-12-24:
XIAO_BLE_Sense has battery charging and battery voltage monitoring functions. I am trying to check these functions with “mbed 2.7.2” or "mbed 2.8.1".
The charging function worked as expected, but “analogRead(P0_31)” to read the battery voltage did not work. To check, I tried other analog pins from A0 to A5, and “analogRead()” worked without any problem.

It seems that there is a problem in the macro “analogPinToPinName(P)” used in the “analogRead()” function. In the definition, “g_APinDescription” (digital pin description) is used instead of “g_AAnalogPinDescription” (analog pin description).
/Arduino15/packages/Seeeduino/hardware/mbed/2.7.2/cores/arduino/wiring_analog.cpp : 89

I changed “g_APinDescription” to “g_AAnalogPinDescription” and now “analogRead(P0_31)” also works as expected.

Is this a bug or a specification?

See also the link below.
https://forum.seeedstudio.com/t/xiao-ble-sense-mbed-2-7-2-battery-charge-and-voltage-monitor-analogread-p0-31-does-not-work/266438

msfujino, 2023-01-05:
Today, mbed 2.9.0 was released, but for some reason this problem has not been resolved. Unfortunately, in order to read the battery voltage, I need to fix /Arduino15/packages/seeduino/hardware/mbed/2.9.0/cores/arduino/pinDefinitions.h : 22.
`//#define analogPinToPinName(P)       (P >= PINS_COUNT ? NC : P < A0 ? g_APinDescription[P+A0].name : g_APinDescription[P].name)
#define analogPinToPinName(P)       (P >= PINS_COUNT ? NC : P < A0 ? g_AAnalogPinDescription[P+A0].name : g_AAnalogPinDescription[P].name)`

msfujino, 2023-01-05:
In mbed 2.9.0, the definition of PIN_VBAT has been changed from (32u) to (6u).
Instead of analogRead(P0_31), analogRead(PIN_VBAT) does not work. In addition, serial monitor does not come up and IDE hangs.

msfujino, 2023-02-08:
v2.9.1 has been released, but
analogRead(P0_31) returns 0xFFFFFFFFFF and analogRead(PIN_VBAT) hangs the IDE.
It is not usable at all.

msfujino, 2023-02-22:
By replacing "variant.cpp" and "pins_arduino.h" in v2.9.1 with my modified "variant.cpp" and "pins_arduino.h", analogRead(P0_31) and analogRead(PIN_VBAT) and green LED will work correctly.
[analogReadL_EDG.zip](https://github.com/Seeed-Studio/ArduinoCore-mbed/files/10799007/analogReadL_EDG.zip)

misterf13, 2023-03-08:
Soooo, is anyone looking into this?

misterf13, 2023-03-08:
I tried the fix from @msfujino and it works great. Could this be merged at some point for future releases?