Seeed-Studio/ArduinoCore-mbed #18: QSPI BlockDevice is not working properly

yangx30, 2023-03-17:
The demo code [https://github.com/Seeed-Studio/ArduinoCore-mbed/blob/master/libraries/Spi_Flash/example/flash_write_read/flash_write_read.ino](url)
does not compile. 

OS is windows 10, installed seeed nrf52 mbed-enabled board v2.9.1 from arduino board manager.
Tried applying patches 0091/0092 approach as described in demo code commit message. Does not work neither. 

(BTW, when running ./mbed-os-to-arduino -r /absolute/path/of/mbed-os/ SEEED_XIAO_NRF52840_SENSE:SEEED_XIAO_NRF52840_SENSE, the requirement.txt file in Local/Temp/mbed-os-program/mbed-os requires pywin32>=301. However, it cannot met by the python27 pip that used in the installing process. I guess this is a mbed cli issue).

The following is the error message.

In file included from C:\Users\yang.x.30\OneDrive - Procter and Gamble\Desktop\arduino\seeed_qspi_test\seeed_qspi_test.ino:7:0:
C:\Users\yang.x.30\AppData\Local\Arduino15\packages\Seeeduino\hardware\mbed\2.9.1\cores\arduino/mbed/storage/blockdevice/COMPONENT_QSPIF/include/QSPIF/QSPIFBlockDevice.h:266:5: error: 'qspi_status_t' does not name a type
     qspi_status_t _qspi_send_program_command(mbed::qspi_inst_t prog_instruction, const void *buffer,
     ^~~~~~~~~~~~~
C:\Users\yang.x.30\AppData\Local\Arduino15\packages\Seeeduino\hardware\mbed\2.9.1\cores\arduino/mbed/storage/blockdevice/COMPONENT_QSPIF/include/QSPIF/QSPIFBlockDevice.h:270:5: error: 'qspi_status_t' does not name a type
     qspi_status_t _qspi_send_read_command(mbed::qspi_inst_t read_instruction, void *buffer, mbed::bd_addr_t addr, mbed::bd_size_t size);
     ^~~~~~~~~~~~~
In file included from C:\Users\yang.x.30\OneDrive - Procter and Gamble\Desktop\arduino\seeed_qspi_test\seeed_qspi_test.ino:7:0:
C:\Users\yang.x.30\AppData\Local\Arduino15\packages\Seeeduino\hardware\mbed\2.9.1\cores\arduino/mbed/storage/blockdevice/COMPONENT_QSPIF/include/QSPIF/QSPIFBlockDevice.h:273:5: error: 'qspi_status_t' does not name a type
     qspi_status_t _qspi_send_erase_command(mbed::qspi_inst_t erase_instruction, mbed::bd_addr_t addr, mbed::bd_size_t size);
     ^~~~~~~~~~~~~
C:\Users\yang.x.30\AppData\Local\Arduino15\packages\Seeeduino\hardware\mbed\2.9.1\cores\arduino/mbed/storage/blockdevice/COMPONENT_QSPIF/include/QSPIF/QSPIFBlockDevice.h:276:5: error: 'qspi_status_t' does not name a type
     qspi_status_t _qspi_send_general_command(mbed::qspi_inst_t instruction_int, mbed::bd_addr_t addr, const char *tx_buffer,
     ^~~~~~~~~~~~~
C:\Users\yang.x.30\AppData\Local\Arduino15\packages\Seeeduino\hardware\mbed\2.9.1\cores\arduino/mbed/storage/blockdevice/COMPONENT_QSPIF/include/QSPIF/QSPIFBlockDevice.h:285:5: error: 'qspi_status_t' does not name a type
     qspi_status_t _qspi_read_status_registers(uint8_t *reg_buffer);
     ^~~~~~~~~~~~~
C:\Users\yang.x.30\AppData\Local\Arduino15\packages\Seeeduino\hardware\mbed\2.9.1\cores\arduino/mbed/storage/blockdevice/COMPONENT_QSPIF/include/QSPIF/QSPIFBlockDevice.h:288:5: error: 'qspi_status_t' does not name a type
     qspi_status_t _qspi_write_status_registers(uint8_t *reg_buffer);
     ^~~~~~~~~~~~~
C:\Users\yang.x.30\AppData\Local\Arduino15\packages\Seeeduino\hardware\mbed\2.9.1\cores\arduino/mbed/storage/blockdevice/COMPONENT_QSPIF/include/QSPIF/QSPIFBlockDevice.h:291:5: error: 'qspi_status_t' does not name a type
     qspi_status_t _qspi_set_frequency(int freq);
     ^~~~~~~~~~~~~
C:\Users\yang.x.30\AppData\Local\Arduino15\packages\Seeeduino\hardware\mbed\2.9.1\cores\arduino/mbed/storage/blockdevice/COMPONENT_QSPIF/include/QSPIF/QSPIFBlockDevice.h:294:5: error: 'qspi_status_t' does not name a type
     qspi_status_t _qspi_update_4byte_ext_addr_reg(mbed::bd_addr_t addr);
     ^~~~~~~~~~~~~
C:\Users\yang.x.30\AppData\Local\Arduino15\packages\Seeeduino\hardware\mbed\2.9.1\cores\arduino/mbed/storage/blockdevice/COMPONENT_QSPIF/include/QSPIF/QSPIFBlockDevice.h:344:11: error: 'QSPI' in namespace 'mbed' does not name a type
     mbed::QSPI _qspi;
           ^~~~
C:\Users\yang.x.30\AppData\Local\Arduino15\packages\Seeeduino\hardware\mbed\2.9.1\cores\arduino/mbed/storage/blockdevice/COMPONENT_QSPIF/include/QSPIF/QSPIFBlockDevice.h:349:12: error: 'SingletonPtr' does not name a type
     static SingletonPtr<PlatformMutex> _devices_mutex;
            ^~~~~~~~~~~~
C:\Users\yang.x.30\AppData\Local\Arduino15\packages\Seeeduino\hardware\mbed\2.9.1\cores\arduino/mbed/storage/blockdevice/COMPONENT_QSPIF/include/QSPIF/QSPIFBlockDevice.h:358:5: error: 'PlatformMutex' does not name a type
     PlatformMutex _mutex;
     ^~~~~~~~~~~~~
C:\Users\yang.x.30\AppData\Local\Arduino15\packages\Seeeduino\hardware\mbed\2.9.1\cores\arduino/mbed/storage/blockdevice/COMPONENT_QSPIF/include/QSPIF/QSPIFBlockDevice.h:361:11: error: 'qspi_inst_t' in namespace 'mbed' does not name a type
     mbed::qspi_inst_t _read_instruction;
           ^~~~~~~~~~~
C:\Users\yang.x.30\AppData\Local\Arduino15\packages\Seeeduino\hardware\mbed\2.9.1\cores\arduino/mbed/storage/blockdevice/COMPONENT_QSPIF/include/QSPIF/QSPIFBlockDevice.h:365:11: error: 'qspi_inst_t' in namespace 'mbed' does not name a type
     mbed::qspi_inst_t _write_status_reg_2_inst;
           ^~~~~~~~~~~
C:\Users\yang.x.30\AppData\Local\Arduino15\packages\Seeeduino\hardware\mbed\2.9.1\cores\arduino/mbed/storage/blockdevice/COMPONENT_QSPIF/include/QSPIF/QSPIFBlockDevice.h:366:11: error: 'qspi_inst_t' in namespace 'mbed' does not name a type
     mbed::qspi_inst_t _read_status_reg_2_inst; // If three registers, this instruction reads the latter two
           ^~~~~~~~~~~
C:\Users\yang.x.30\AppData\Local\Arduino15\packages\Seeeduino\hardware\mbed\2.9.1\cores\arduino/mbed/storage/blockdevice/COMPONENT_QSPIF/include/QSPIF/QSPIFBlockDevice.h:371:11: error: 'qspi_inst_t' in namespace 'mbed' does not name a type
     mbed::qspi_inst_t _4byte_msb_reg_write_inst;
           ^~~~~~~~~~~
C:\Users\yang.x.30\AppData\Local\Arduino15\packages\Seeeduino\hardware\mbed\2.9.1\cores\arduino/mbed/storage/blockdevice/COMPONENT_QSPIF/include/QSPIF/QSPIFBlockDevice.h:392:5: error: 'qspi_bus_width_t' does not name a type
     qspi_bus_width_t _inst_width; //Bus width for Instruction phase
     ^~~~~~~~~~~~~~~~
C:\Users\yang.x.30\AppData\Local\Arduino15\packages\Seeeduino\hardware\mbed\2.9.1\cores\arduino/mbed/storage/blockdevice/COMPONENT_QSPIF/include/QSPIF/QSPIFBlockDevice.h:393:5: error: 'qspi_bus_width_t' does not name a type
     qspi_bus_width_t _address_width; //Bus width for Address phase
     ^~~~~~~~~~~~~~~~
C:\Users\yang.x.30\AppData\Local\Arduino15\packages\Seeeduino\hardware\mbed\2.9.1\cores\arduino/mbed/storage/blockdevice/COMPONENT_QSPIF/include/QSPIF/QSPIFBlockDevice.h:394:5: error: 'qspi_address_size_t' does not name a type; did you mean 'nrf_twi_address_set'?
     qspi_address_size_t _address_size; //Number of bits for address
     ^~~~~~~~~~~~~~~~~~~
     nrf_twi_address_set
C:\Users\yang.x.30\AppData\Local\Arduino15\packages\Seeeduino\hardware\mbed\2.9.1\cores\arduino/mbed/storage/blockdevice/COMPONENT_QSPIF/include/QSPIF/QSPIFBlockDevice.h:395:5: error: 'qspi_alt_size_t' does not name a type; did you mean 'pin_size_t'?
     qspi_alt_size_t _alt_size; //Number of bits for alt
     ^~~~~~~~~~~~~~~
[... 25 lines cut ...]
C:\Users\yang.x.30\OneDrive - Procter and Gamble\Desktop\arduino\seeed_qspi_test\seeed_qspi_test.ino:11:74: error: 'QSPI_FLASH1_IO3' was not declared in this scope
 QSPIFBlockDevice root(QSPI_FLASH1_IO0, QSPI_FLASH1_IO1, QSPI_FLASH1_IO2, QSPI_FLASH1_IO3, QSPI_FLASH1_SCK, QSPI_FLASH1_CSN, QSPIF_POLARITY_MODE_1, MBED_CONF_QSPIF_QSPI_FREQ);
                                                                          ^~~~~~~~~~~~~~~
C:\Users\yang.x.30\OneDrive - Procter and Gamble\Desktop\arduino\seeed_qspi_test\seeed_qspi_test.ino:11:74: note: suggested alternative: 'QSPI_PIN_IO3'
 QSPIFBlockDevice root(QSPI_FLASH1_IO0, QSPI_FLASH1_IO1, QSPI_FLASH1_IO2, QSPI_FLASH1_IO3, QSPI_FLASH1_SCK, QSPI_FLASH1_CSN, QSPIF_POLARITY_MODE_1, MBED_CONF_QSPIF_QSPI_FREQ);
                                                                          ^~~~~~~~~~~~~~~
                                                                          QSPI_PIN_IO3
C:\Users\yang.x.30\OneDrive - Procter and Gamble\Desktop\arduino\seeed_qspi_test\seeed_qspi_test.ino:11:91: error: 'QSPI_FLASH1_SCK' was not declared in this scope
 QSPIFBlockDevice root(QSPI_FLASH1_IO0, QSPI_FLASH1_IO1, QSPI_FLASH1_IO2, QSPI_FLASH1_IO3, QSPI_FLASH1_SCK, QSPI_FLASH1_CSN, QSPIF_POLARITY_MODE_1, MBED_CONF_QSPIF_QSPI_FREQ);
[... cut ...]
 QSPIFBlockDevice root(QSPI_FLASH1_IO0, QSPI_FLASH1_IO1, QSPI_FLASH1_IO2, QSPI_FLASH1_IO3, QSPI_FLASH1_SCK, QSPI_FLASH1_CSN, QSPIF_POLARITY_MODE_1, MBED_CONF_QSPIF_QSPI_FREQ);
                                                                                                            ^~~~~~~~~~~~~~~
                                                                                                            QSPI_PIN_CSN

exit status 1

Compilation error: 'QSPI_FLASH1_IO0' was not declared in this scope

yangx30, 2023-03-23:
need to checkout the latest commit after  v2.9.1 release

closed 2023-03-23 by yangx30

josephduchesne, 2023-11-09:
This is caused by a regression in the compiled libmbed.a.

I used the old version from here:
https://github.com/Seeed-Studio/ArduinoCore-mbed/commit/4a6a255c9f00abe9ca051aebdbce734d784cc024

(Which was compiled with QSPI support) and it seems to work fine, but the versions available in this repo's releases are all missing the QSPI symbols.